devWaves/SwitchBot-MQTT-BLE-ESP32 #113: FingerBot support?

andypnz, 2022-11-22:
Hi. Do you think there is any way to modify this to work with an Adaprox FingerBot?

I have your MQTT / ESP32 working with HA.

ESPHome ESP32_BLE_Tracker gives this on a FingerBot:

``
[16:18:49][D][ble_client.automation:048]: Connection established with dc:23:xx:xx:xx:xx [16:18:49][I][esp32_ble_client:154]: [0] [dc:23:xx:xx:xx:xx] 
Service UUID: 0x1800 [16:18:49][I][esp32_ble_client:156]: [0] [dc:23:xx:xx:xx:xx] start_handle: 0x1 end_handle: 0x3 [16:18:49][I][esp32_ble_client:059]: [0] [dc:23:4d:ae:25:c9] 
characteristic 0x2A00, handle 0x3, properties 0xa [16:18:49][I][esp32_ble_client:154]: [0] [dc:23:xx:xx:xx:xx] 
Service UUID: 0x1801 [16:18:49][I][esp32_ble_client:156]: [0] [dc:23:xx:xx:xx:xx] start_handle: 0x4 end_handle: 0x7 [16:18:49][I][esp32_ble_client:059]: [0] [dc:23:xx:xx:xx:xx] 
characteristic 0x2A05, handle 0x6, properties 0x22 [16:18:49][I][esp32_ble_client:154]: [0] [dc:23:xx:xx:xx:xx] 
Service UUID: 0x1910 [16:18:49][I][esp32_ble_client:156]: [0] [dc:23:4d:ae:25:c9] start_handle: 0x8 end_handle: 0xd [16:18:50][I]esp32_ble_client:059]: [0] [dc:23:xx:xx:xx:xx] 
characteristic 0x2B11, handle 0xa, properties 0x4 [16:18:50][I][esp32_ble_client:059]: [0] [dc:23:xx:xx:xx:xx] characteristic 0x2B10, handle 0xc, properties 0x10
``
Thanks!

garide78, 2022-12-06:
good evening

interested too, if FINGERBOTS could be integrated

News?

thank you

devWaves, 2022-12-07:
the fingerbot most likely does not work the exact same and would take significant work to integrate

also I dont believe the fingerbot BLE commands are available also without some reverse engineering