devWaves/SwitchBot-MQTT-BLE-ESP32 #116: Switchbot found but further scanning forever

infourbonsnet, 2022-12-21:
I have configured 1 switchbot device. Device is found during initial scan, but scan is continued afterwards forever. Here is the output from 2 scan cycles:

> START onResult
> Advertised Device found: 
> Name: TY, Address: dc:23:4e:9f:d3:4b, manufacturer data: d007800300000100fc99e0d97d4c2ff6df47d2cbbfb314f2, serviceUUID: 0xa201
> Service Data:
> UUID: 0xa201, Data: 
> Adding Our Service ... 
> switchbotone
> START callForInfoAdvDev
> callForInfoAdvDev
> END onResult
> START onResult
> Advertised Device found: 
> Name: TY, Address: dc:23:4e:9f:d3:4b, manufacturer data: d007800300000100fc99e0d97d4c2ff6df47d2cbbfb314f2, serviceUUID: 0xa201
> Service Data:
> UUID: 0xa201, Data: 
> Adding Our Service ... 
> switchbotone

Through adding additional traces I have identified the highlighted branch being responsible for continued scan:

    bool callForInfoAdvDev(std::string deviceMac, long anRSSI,  std::string & aValueString) {
      yield();
      printAString("START callForInfoAdvDev");
      printAString("callForInfoAdvDev");
      Serial.printf("deviceMac: %s, anRSSI: %d, aValueString: %s\n", deviceMac.c_str(), anRSSI, aValueString.c_str());
      if ((strcmp(deviceMac.c_str(), "") == 0)) {
          return false;
      }
      if ((strcmp(aValueString.c_str(), "") == 0)) {
      // function is returning from HERE
        return false;
      }

At this point, "aValueString" seems to represent:
                  aValueString = advertisedDevice->getManufacturerData();

Acc. to the debug output, it is actually not empty.

Any advice is very welcome.

devWaves, 2022-12-22:
that doesnt look like an official switchbot

the name TY leads me to believe it is a TUYA device

infourbonsnet, 2022-12-22:
Well, it's not an offial switchbot. I've actually not been aware that compatibility is limited to those devices. But if that's the case - understood.
Do you have any advice how to make a Tuya Finger Bot accessible through MQTT?

infourbonsnet, 2022-12-23:
Thanks anyway!

closed 2022-12-23 by infourbonsnet