devWaves/SwitchBot-MQTT-BLE-ESP32 #119: Firmware update curtain

Capusjon, 2023-03-12:
There seems to be a firmware update for the switchbot curtain. (See screenshot)

Is it safe to proceed with the update or should we withhold updating the curtain bots?

Kund regards
Colin
[image]

ReGGaeGaNdi, 2023-03-22:
Hi Colin
I'm using firmware version 6.0 since a while already and everything still is working fine, so I think you also should be able to upgrade without issues.
Regards GaNdi

devWaves, 2023-03-23:
Yes v6.0 for curtain should work

Capusjon, 2023-03-24:
Thanks for the response!

aptalca, 2023-03-28:
Was the firmware pulled? I did get the notice for the v6 upgrade being available and I decided not to do it until I got confirmation here.

Now the switchbot app no longer shows me the v6 upgrade as available 😕