devWaves/SwitchBot-MQTT-BLE-ESP32 #120: SwitchBot-MQTT-BLE-ESP32 v7.1 request the IP to DHCP every second in loop

iz8mbw, 2023-03-17:
Hello.
Using SwitchBot-MQTT-BLE-ESP32 v7.1 on my ESP32 board.
The ESP32 requests in loop every 1 second the IP to DHCP Server.

See here the DHCP log:

`
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 available DHCP range: 192.168.1.2 -- 192.168.1.254
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 client provides name: esp32
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 DHCPDISCOVER(eth0) 94:3c:c6:xx:xx:xx
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 tags: known, eth0
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 DHCPOFFER(eth0) 192.168.1.32 94:3c:c6:xx:xx:xx
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server,
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 requested options: 42:ntp-server
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 next server: 192.168.1.2
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  1 option: 53 message-type  2
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 54 server-identifier  192.168.1.2
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 51 lease-time  12h
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 58 T1  6h
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 59 T2  10h30m
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option:  1 netmask  255.255.255.0
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 28 broadcast  192.168.1.255
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  8 option:  6 dns-server  192.168.1.4, 192.168.1.2
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 42 ntp-server  193.204.114.105
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option:  3 router  192.168.1.1

Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 available DHCP range: 192.168.1.2 -- 192.168.1.254
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 client provides name: esp32
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 DHCPREQUEST(eth0) 192.168.1.32 94:3c:c6:xx:xx:xx
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 tags: known, eth0
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 DHCPACK(eth0) 192.168.1.32 94:3c:c6:xx:xx:xx esp32
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server,
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 requested options: 42:ntp-server
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 next server: 192.168.1.2
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  1 option: 53 message-type  5
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 54 server-identifier  192.168.1.2
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 51 lease-time  12h
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 58 T1  6h
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 59 T2  10h30m
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option:  1 netmask  255.255.255.0
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 28 broadcast  192.168.1.255
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  8 option:  6 dns-server  192.168.1.4, 192.168.1.2
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option: 42 ntp-server  193.204.114.105
Mar 17 08:37:34 dnsmasq-dhcp[638]: 1939379725 sent size:  4 option:  3 router  192.168.1.1

Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 available DHCP range: 192.168.1.2 -- 192.168.1.254
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 client provides name: esp32
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 DHCPDISCOVER(eth0) 94:3c:c6:xx:xx:xx
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 tags: known, eth0
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 DHCPOFFER(eth0) 192.168.1.32 94:3c:c6:xx:xx:xx
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server,
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 requested options: 42:ntp-server
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 next server: 192.168.1.2
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  1 option: 53 message-type  2
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  4 option: 54 server-identifier  192.168.1.2
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  4 option: 51 lease-time  12h
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  4 option: 58 T1  6h
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  4 option: 59 T2  10h30m
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  4 option:  1 netmask  255.255.255.0
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  4 option: 28 broadcast  192.168.1.255
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  8 option:  6 dns-server  192.168.1.4, 192.168.1.2
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  4 option: 42 ntp-server  193.204.114.105
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 sent size:  4 option:  3 router  192.168.1.1

Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 available DHCP range: 192.168.1.2 -- 192.168.1.254
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 client provides name: esp32
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 DHCPREQUEST(eth0) 192.168.1.32 94:3c:c6:xx:xx:xx
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 tags: known, eth0
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 DHCPACK(eth0) 192.168.1.32 94:3c:c6:xx:xx:xx esp32
Mar 17 08:37:35 dnsmasq-dhcp[638]: 4060657655 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server,
[... 70 lines cut ...]
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 available DHCP range: 192.168.1.2 -- 192.168.1.254
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 client provides name: esp32
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 DHCPREQUEST(eth0) 192.168.1.32 94:3c:c6:xx:xx:xx
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 tags: known, eth0
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 DHCPACK(eth0) 192.168.1.32 94:3c:c6:xx:xx:xx esp32
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server,
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 requested options: 42:ntp-server
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 next server: 192.168.1.2
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  1 option: 53 message-type  5
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  4 option: 54 server-identifier  192.168.1.2
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  4 option: 51 lease-time  12h
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  4 option: 58 T1  6h
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  4 option: 59 T2  10h30m
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  4 option:  1 netmask  255.255.255.0
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  4 option: 28 broadcast  192.168.1.255
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  8 option:  6 dns-server  192.168.1.4, 192.168.1.2
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  4 option: 42 ntp-server  193.204.114.105
Mar 17 08:37:38 dnsmasq-dhcp[638]: 175962372 sent size:  4 option:  3 router  192.168.1.1

`

as you can understand this is not a normal behavior since the DHCP correctly provide the IP address in fact I can reach the ESP32 using the IP that the DHCP provided.
This behavior generate lots of useless traffic in WiFi and I noticed also that the board gets hot.

Thanks @devWaves

devWaves, 2023-03-23:
hey. I'm not sure why it would be doing that, mine does not

Are you by chance using a meshed router/network? if so, can you bind the esp32 to one access point?

also what type of esp32 board are you using?

iz8mbw, 2023-03-23:
Hi @devWaves 
I did some more test and this DHCP loop issue appears when the MQTT Server is offline.
So to reproduce/simulate this DHCP loop issue simply stop the MQTT Server and you will se the ESP32 is going to made DHCP requests in loop.

When MQTT Server is UP this DHCP loop issue disappears.

I understand that SwitchBot-MQTT-BLE-ESP32 v7.1 wants a reachable MQTT Server but in case of MQTT Server failure it's not normal that SwitchBot-MQTT-BLE-ESP32 v7.1 starts to made DHCP requests in loop.

Thanks.