devWaves/SwitchBot-MQTT-BLE-ESP32 #123: No device

superino1, 2023-03-31:
Hi all, I tried to add my switchbot, but it doesn't add.  I have downloaded all the libraries mentioned, I only see the WiFi signal esp32 sensor.

ahori, 2023-04-01:
I had the same problem.  In my case, the 'mqtt_main_topic' was left unchanged ('switchbot'). When I changed it to 'homeassistant', then it was fixed.

superino1, 2023-04-01:
Thank you, but for me, no work :(

iz8mbw, 2023-04-02:
Did you edit the .ino file with the Mac Address of your Switchbot device?

superino1, 2023-04-02:
> Did you edit the .ino file with the Mac Address of your Switchbot device?

Yes, I followed the guide

ahori, 2023-04-02:
Well, did you check the MQTT broker status by using MQTT Explorer app?  If so, may I check its output?

superino1, 2023-04-02:
> Well, did you check the MQTT broker status by using MQTT Explorer app? If so, may I check its output?

With mqtt explorer I only see esp32, no switchbot

ahori, 2023-04-02:
Here is mine;
[MQTT-Explorer-Screenshot.pdf](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/files/11131977/MQTT-Explorer-Screenshot.pdf)

I guess your Bluetooth MAC addresses of switchbot devices are wrong and Switchbot-MQTT... cannot find switchbot devices.

superino1, 2023-04-02:
[image]

> 
> I guess your Bluetooth MAC addresses of switchbot devices are wrong and Switchbot-MQTT... cannot find switchbot devices.

My mistake, sorry, I didn't delete *.. Now I see it, but it doesn't work, it didn't take any commands and after a while it went to unavailable.
Have switchbot one