devWaves/SwitchBot-MQTT-BLE-ESP32 #125: No Wifi connection

MrMoblys, 2023-04-30:
Good morning,

I bought a D1 Mini ESP32 (exactly ESP32-WROOM-32) from AZ-Delivery and when I fill in my IDs for Wifi, there is no connection.
Are there any specific prerequisites for the WiFi settings?
I tried SSID hidden or not, without special characters, on another terminal, near or far from the terminal... nothing works.

Or did I mess up?

Thanks

devWaves, 2023-05-01:
Is your WIFI 2.4 Ghz? It won't work with 5 Ghz

You can try with a simple ESP32 wifi sketch to test the wifi

MrMoblys, 2023-05-01:
Hello,

The Wifi is 2.4Ghz and tested with a Wifi test sketch.
I have an ESP8266 lying around and I put the same type of sketch and it connects.
I tried going through ESPHome and it blocks the Wifi connection. The scan is possible but not the connection.
I think the problem comes from my ESP32. I am in contact with AZ-Delivery for the return of the ESP.

Sorry to bother you

DrRSatzteil, 2023-05-08:
I use the exact same model from AZ delivery without any issues so I guess you should have this unit replaced by them.

MrMoblys, 2023-05-08:
Thank you for your return.

Indeed, that's what I'm going to do. I'm waiting for the ESP32 and I'll get back to you ;-)

MrMoblys, 2023-05-10:
Hello

Changing the ESP solved my problem. Thanks for your help.

closed 2023-05-10 by MrMoblys