devWaves/SwitchBot-MQTT-BLE-ESP32 #126: Opening one side of the curtain

haroldboom, 2023-05-14:
Hi, firstly thanks for the integration it works fantastic.

However I was wondering if it is possible to open just one side of a dual curtain setup? My switchbots are grouped and added as a single Bluetooth MAC address. Is it possible to just open one side in this configuration or do I need to ungrouo them and add them individually?

thanks

devWaves, 2023-05-17:
in order to do that you would need to break them up as 2 devices

You would then send 2 seperate commands from the esp32, one to each curtain

when linked, it works a similar way but the esp32 would send the command to the primary curtain, then the curtain would send the second command to the other curtain

depending on the distance of the esp32 to the curtains you can get a delay on the second command that could be greater then the delay of the primary curtain sending a command to the 2nd curtain

haroldboom, 2023-05-17:
Yeah no worries I should be able to make that work. Thanks again mate