devWaves/SwitchBot-MQTT-BLE-ESP32 #132: SwitchBot Blind tilt

tim354, 2023-07-04:
Hello,
does the new SwitchBot Blind tilt also work with this Project?

zolakk, 2023-07-05:
Not currently. I've tried to hack it in on [this issue](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/117#issuecomment-1507582412) but got stuck and over my head with my programming abilities. If someone more capable were able to take my findings and go that last mile so to speak it would work. I feel like it's something simple I'm missing but have no idea what it could be but for the time being it's stalled.

brianharwell, 2023-10-27:
Plus 1 for this!

I bought 3 blind tilts thinking they would work but now I need to return them. :-(

zolakk, 2023-10-28:
If you use Home Assistant the native integration works just fine over Bluetooth. It's actually one of the only reasons I now run HA along side OpenHAB to get that control. I've yet to figure out how to surface the blind controlled by HA into OH effectively though so that's annoying

brianharwell, 2023-10-28:
Yea I use HA. My NUC is my office closet and I bought a Bluetooth adapter, a USB hub, and a 6ft extension cable to see if this will work.

I was SOOOO hoping to use your app. I even have the ESP32 setup…then I realized no blind tilt support.

What skill set would be required to add this functionality?

zolakk, 2023-10-28:
Really it's just the ability to understand the C (I think) this is written in and look at the python the HA integration uses for the blind Tilt and be able to translate it. You can probably get away with duplicating most of the code sections here for curtain, it's just the parts where the it decodes the bits for status and stuff that are different. I got part of the way there in the issue I linked above but couldn't quite get it to work. It's probably just a few lines of code that need to change but try as I might I just couldn't get it to work