devWaves/SwitchBot-MQTT-BLE-ESP32 #133: Issue with adding SwitchBot meter: Outdoor Hygrometer Thermometer

bklen, 2023-07-09:
I am having some issues with adding a switchbot "Thermo-Hygrometer", and hoping you all can help me out!  

**Sensor**: SwitchBot IP65 Indoor Outdoor Hygrometer Thermometer  
amazon link: https://a.co/d/9VOZyo0  
SwitchBot link: https://us.switch-bot.com/products/switchbot-indoor-outdoor-thermo-hygrometer?variant=44560794812649  

&nbsp;
I am able to add the sensor to Home Assistant using the SwitchBot add-on: https://drive.google.com/file/d/1pfbvTVE9Trey5ngG6uuW4FvJw8GnbKeH/view?usp=drive_link  

&nbsp;
However, after following the instructions(https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32), I am having issues: https://drive.google.com/file/d/1Nlg9fyvSreWnqrhoBg-ZQtTUx0pQ9evb/view?usp=drive_link
* Checking MQTT Explorer, the meter shows up under "homeassistant", and not under "switchbot"  
* Sensor values show up as "Unknown"  

&nbsp;
I tried adding the sensor as a "Bot", just to see what would happen: https://drive.google.com/file/d/1UePuTuZMshxPls3sqSwUzsE-OLbFpe20/view?usp=drive_link
* MQTT Explorer shows the bot in "switchbot" topic and "homeassistant" topic.  
* I do see "Battery", "Linkquality", and "Mode". However, I obviously can't see "temperature" or "humidity"  
* Excuse my ignorance as I am new to MQTT, but how come the added sensor/device shows up under the "homeassistant" topic. I assumed it would just show up under the created "switchbot"  mqtt_main_topic  

&nbsp;  
Please let me know if you need any more information, and thank you!!!

Nardol, 2023-07-16:
What other Switch Bot devices do you have?

Because you could try to use passive scan and see if outdoor meters works, as the way to retrieve their data is different than the one for classic meters in active mode.
The problem is, if you have I.E. bot or curtains these devices won't work anymore.
But if you can try temporarily to switch to passive scan it could help to see if the issue is the one I suggest.

Morpheus2018, 2023-07-16:
#### I'm having the same problems with that "SwitchBot IP65 Indoor Outdoor Hygrometer Thermometer "

[image]
[image]

#### my settings in the SwitchBot-BLE2MQTT-ESP32.ino

```
/* Wifi Settings */
static const char* host = "esp32";                                  //  Unique name for ESP32. The name detected by your router and MQTT. If you are using more then 1 ESPs to control different switchbots be sure to use unique hostnames. Host is the MQTT Client name and is used in MQTT topics
static const char* ssid = "MY-WLAN";                                   //  WIFI SSID
static const char* password = "7817931Z";                           //  WIFI Password

/* MQTT Settings */
/* MQTT Client name is set to WIFI host from Wifi Settings*/
static const char* mqtt_host = "192.168.178.99";                       //  MQTT Broker server ip
static const char* mqtt_user = "mqtt-user";                         //  MQTT Broker username. If empty or NULL, no authentication will be used
static const char* mqtt_pass = "x7#S@9";                         //  MQTT Broker password
static const int mqtt_port = 1883;                                  //  MQTT Port
static const std::string mqtt_main_topic = "switchbot";             //  MQTT main topic

/* Mesh Settings */
/* Ignore if only one ESP32 is used */
.........
.....
...

/* Switchbot Meter Settings */
static std::map<std::string, std::string> allMeters = {
    { "meterone", "C9:0E:7F:17:4E:EA" }/* ,
    { "metertwo", "yY:yY:yY:yY:yY:yY" }*/
};
..........
......
... 
```

RobusX, 2023-07-21:
Hi,

I can confirm the same issue in my case.
I bought three "SwitchBot IP65 Indoor Outdoor Hygrometer Thermometer" for outside space (without LCD display) and  two "SwitchBot Indoor Thermometer Hygrometer" (with LCD display). Via SwitchBot-MQTT-BLE-ESP32 integration inside Mosquitto MQTT are "normally" displayed only two inside thermometers (with LCD), other three for outside have all "unknown" values for temperature, humidity, link quality, battery. Within MQTT explorer are visible only two with LCDs.
But, all thermometers and their values are normally visible and operational with Switchbot integration via BLEProxy (separate ESP32 board). So, there should be some issue within  "SwitchBot-MQTT-BLE-ESP32" for "SwitchBot IP65 Indoor Outdoor Hygrometer Thermometer" . Thanks in advance.

Nardol, 2023-07-21:
Has someone tested what I [suggested above](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/133#issuecomment-1637042751)?
Because it could give an idea about what is causing this issue.

bklen, 2023-07-21:
@Nardol I've been really busy lately and haven't had the time to give your a suggestion a try yet. In the meantime, I did the same thing the previous poster mentioned. I set up a Bluetooth proxy with esphome and used the switchbot integration.

Nardol, 2023-07-21:
@bklen thanks for your reply.
If you only have thermometers, IMO the native HA integration is sufficient except if I missed something.
Most advantages of using switchbot MQTT BLE Esp32 are for devices with interactions as Bot or curtains for example.

RobusX, 2023-07-21:
@Nardol 
I set "onlyPassiveScan = true" and compiled/uploaded to my ESP32 Wroom ("ESP32dev" with PlatformIO).
Now in HA all three outside thermometers  have visible values (humidity, temperature, link quality), except battery state is still "unknown" (also battery state is not visible with  mqtt explorer for three outside thermometers, only inside thermometers have battery data ttransferred via mqtt).
I will keep both Switchbot and SwitchBot-MQTT-BLE-ESP32 integrations for redundancy (due to possibility or eventually failure one of them with HA upgrade). 
And possibly SwitchBot-MQTT-BLE-ESP32 has also faster updates values within HA (it seems so).
Thank you very much for your advice. Respect.

Nardol, 2023-07-22:
Oh, strange that battery information are displayed for inside thermometers.
Or it is because previous values are still retained by your MQTT broker, If I am right battery state cannot be retrieved in passive scan mode.

Morpheus2018, 2023-07-28:
> 

After the changes "static const bool onlyPassiveScan = true;" the battery status is "Unknown" !!!

```
static const bool onlyActiveScan = false;                    // Active scanning requires more battery from the BLE switchbot devices. If false, passive scanning is used when possible for contact/motion
static const bool onlyPassiveScan = true;                   // If this ESP32 is a mesh ESP32 or you only have motion/contact/meter sensors. Passive scanning uses less battery from BLE switchbot devices. Passive scanning provides less data then active scanning, but uses less battery
static const bool alwaysActiveScan = false;                  // No battery optimizations. If you are using the switchbot hub or app to control devices also and you want immediate state updates for bot and curtains in MQTT set to true
static const bool scanWhileCurtainIsMoving = true;           // The ESP32 will scan for defaultCurtainScanAfterControlSecs seconds after control to keep the position slider in sync with the actual position

```

[image]

Nardol, 2023-07-28:
> After the changes "static const bool onlyPassiveScan = true;" the battery status is "Unknown" !!!

That is expected.
It looks like battery information is not retrieved in passive scan because not part of advertized manufacturer data, only service data provide it and these data are only retrieved in active scan.

The Python implementation also do not provide battery information in passive scan if I remember correctly.

RobusX, 2023-07-28:
No, although passive mode for about one week now, I still see battery status via Mqtt for both  "SwitchBot Indoor Thermometer Hygrometers" (currently 100%). For all three outside thermometers status is "Uknown", as you predicted. Details in pictures below.
[image]
[image]

MichaelBMF, 2023-12-01:
https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/blob/latest/devicetypes/meter.md#outdoor-temperaturehumidity-sensor