devWaves/SwitchBot-MQTT-BLE-ESP32 #134: Warnings when using with Home Assistant 2023.8.0

pickonedev, 2023-08-03:
Hi!

I was using SwitchBot-MQTT-BLE-ESP32 with Home Assistant and till now everything was ok, no error, no warning. From yesterday, when I updated Home Assistant to a newer version, I got this warning, and not only me. Any idea how to solve this? Thank you!

[image]

Nardol, 2023-08-03:
These warning can be safely ignored by users but this project should be updated to reflect changes introduced in HA 2023.8.

@devWaves do you think you could find some time to update names?
I might try to make a PR but I am not sure to know the code enough to do it without breaking something and I personally use Switch Bot with the native integration which is enough for my use so not sure to succeed in making efficient tests.

Morpheus2018, 2023-08-04:
Is there really no need for action and the message can be ignored!?

[image]

pickonedev, 2023-08-04:
At least till 2024.2.0 :-D

[image]

pickonedev, 2023-08-05:
18 warnings in less than 2 minutes

```
Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:294
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 23:40:24 (18 occurrences)
Last logged: 23:40:24

    MQTT entity name starts with the device name in your config {'name': 'door_sensor BinaryContact', 'device': {'identifiers': ['switchbot_DB:63:F6:1C:E1:E2'], 'manufacturer': 'WonderLabs SwitchBot', 'model': 'Contact', 'name': 'door_sensor', 'connections': []}, 'availability_topic': 'switchbot/entrance_switchbot/lastwill', 'unique_id': 'switchbot_DB:63:F6:1C:E1:E2_bincontact', 'icon': 'mdi:door', 'state_topic': 'switchbot/entrance_switchbot/contact/door_sensor/bin', 'payload_on': 'OPEN', 'payload_off': 'CLOSED', 'qos': 0, 'force_update': False, 'availability_mode': 'latest', 'enabled_by_default': True, 'payload_not_available': 'offline', 'encoding': 'utf-8', 'payload_available': 'online'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'BinaryContact'
    MQTT entity name starts with the device name in your config {'name': 'door_sensor In', 'device': {'identifiers': ['switchbot_DB:63:F6:1C:E1:E2'], 'manufacturer': 'WonderLabs SwitchBot', 'model': 'Contact', 'name': 'door_sensor', 'connections': []}, 'availability_topic': 'switchbot/entrance_switchbot/lastwill', 'unique_id': 'switchbot_DB:63:F6:1C:E1:E2_in', 'icon': 'mdi:motion-sensor', 'state_topic': 'switchbot/entrance_switchbot/contact/door_sensor/in', 'payload_on': 'ENTERED', 'payload_off': 'IDLE', 'qos': 0, 'force_update': False, 'availability_mode': 'latest', 'enabled_by_default': True, 'payload_not_available': 'offline', 'encoding': 'utf-8', 'payload_available': 'online'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'In'
    MQTT entity name starts with the device name in your config {'name': 'door_sensor Out', 'device': {'identifiers': ['switchbot_DB:63:F6:1C:E1:E2'], 'manufacturer': 'WonderLabs SwitchBot', 'model': 'Contact', 'name': 'door_sensor', 'connections': []}, 'availability_topic': 'switchbot/entrance_switchbot/lastwill', 'unique_id': 'switchbot_DB:63:F6:1C:E1:E2_out', 'icon': 'mdi:exit-run', 'state_topic': 'switchbot/entrance_switchbot/contact/door_sensor/out', 'payload_on': 'EXITED', 'payload_off': 'IDLE', 'qos': 0, 'force_update': False, 'availability_mode': 'latest', 'enabled_by_default': True, 'payload_not_available': 'offline', 'encoding': 'utf-8', 'payload_available': 'online'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Out'
    MQTT entity name starts with the device name in your config {'name': 'door_sensor Button', 'device': {'identifiers': ['switchbot_DB:63:F6:1C:E1:E2'], 'manufacturer': 'WonderLabs SwitchBot', 'model': 'Contact', 'name': 'door_sensor', 'connections': []}, 'availability_topic': 'switchbot/entrance_switchbot/lastwill', 'unique_id': 'switchbot_DB:63:F6:1C:E1:E2_button', 'state_topic': 'switchbot/entrance_switchbot/contact/door_sensor/button', 'icon': 'mdi:gesture-tap-button', 'payload_on': 'PUSHED', 'payload_off': 'IDLE', 'qos': 0, 'force_update': False, 'availability_mode': 'latest', 'enabled_by_default': True, 'payload_not_available': 'offline', 'encoding': 'utf-8', 'payload_available': 'online'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Button'
    MQTT entity name starts with the device name in your config {'name': 'door_sensor Illuminance', 'device': {'identifiers': ['switchbot_DB:63:F6:1C:E1:E2'], 'manufacturer': 'WonderLabs SwitchBot', 'model': 'Contact', 'name': 'door_sensor', 'connections': []}, 'availability_topic': 'switchbot/entrance_switchbot/lastwill', 'unique_id': 'switchbot_DB:63:F6:1C:E1:E2__illuminance', 'state_topic': 'switchbot/entrance_switchbot/contact/door_sensor/illuminance', 'device_class': <BinarySensorDeviceClass.LIGHT: 'light'>, 'payload_on': 'BRIGHT', 'payload_off': 'DARK', 'qos': 0, 'force_update': False, 'availability_mode': 'latest', 'enabled_by_default': True, 'payload_not_available': 'offline', 'encoding': 'utf-8', 'payload_available': 'online'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Illuminance'
```

gui28347, 2023-08-06:
I was able to fix, several lines need to be changed:
Basically: 

Search for "addToPublish"
    The ones that have line similar with:    + "\"name\":\"" + host + " Linkquality\"," +
                                     Need to be replaced with:    + "\"name\":\"Linkquality\"," +

    The same applies for all other properties like Firmware, Switch ........

 Basically, the change requires that properties only contain property name not deviceName + property name like the code is now.

It would be nice if some developer with access to this repo apply changes and release a version with the fix in it. Really simple changes.

Morpheus2018, 2023-08-07:
> I was able to fix, several lines need to be changed: Basically:
>
> Search for "addToPublish" The ones that have line similar with: + ""name":"" + host + " Linkquality"," + Need to be replaced with: + [...]

which line in the code exactly, do you mean, this [+ "\"name\":\"" + host + " Linkquality\"," +](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3153)  line?

replace with  that? → (  + "\"name\":\" Linkquality\"," + )

also the lines with [+ "\"name\":\"" + deviceName + " Energy\"," +](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3201)........ need to be replaced?

!are not displayed here! " \ "

pickonedev, 2023-08-12:
> 

It is working for you? I still get the same error in Home Assistant...

yauyauwind, 2023-08-24:
Update to HA 2023.8.3 not work , I use the curtain, and ESP32 can detect and work with MQTT, but curtain is not work showing unavailable, roll back to 2023.8.2 is work fine, anyone who updated to 2023.8.3 is work??

yauyauwind, 2023-08-25:
Just update , HA 2023.8.4 Work

iz8mbw, 2023-09-17:
Same issues here (this issue affect all users).

```
WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'name': 'esp32 Firmware', 'device': {'identifiers': ['switchbotesp_esp32_00:3C:C6:37:XX:X0'], 'manufacturer': 'WonderLabs SwitchBot', 'model': 'ESP32', 'name': 'esp32', 'connections': []}, 'availability_topic': 'switchbot/esp32/lastwill', 'unique_id': 'switchbotesp_esp32_00:3C:C6:37:XX:X0_firmware', 'icon': 'mdi:cog', 'state_topic': 'switchbot/esp32/esp32/firmware', 'enabled_by_default': True, 'encoding': 'utf-8', 'payload_not_available': 'offline', 'qos': 0, 'payload_available': 'online', 'force_update': False, 'availability_mode': 'latest'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Firmware'
```

Hope @devWaves will fix it.

Thanks.

pickonedev, 2023-10-15:
Hei... don't wait for devWaves, he will not update this anymore, I am 100% sure.... Instead this, I have another solution, if you use esp32 or similar devices... Try to use ESPHome, and transform your device into an Active Bluetooth device, now Home Assistant will recognize all the switchbot devices, by extending bluetooth range with proxies. I use this code and I can detect and use ALL my switchbot or other bluetooth device...

```

esphome:
  name: XXXX
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
    
captive_portal:
logger:
api:
  encryption:
    key: "XXXX"
ota:

esp32_ble_tracker:
  scan_parameters:
    active: true
bluetooth_proxy:   
  active: true

button:
  - platform: safe_mode
    name: ${display_name} (Safe Mode)
```

yauyauwind, 2023-10-16:
> Hei... don't wait for devWaves, he will not update this anymore, I am 100% sure.... Instead this, I have another solution, if you use esp32 or similar devices... Try to use ESPHome, and transform your device into an Active Bluetooth device, now Home Assistant will recognize all the switchbot devices, by extending bluetooth range with proxies. I use this code and I can detect and use ALL my switchbot or other bluetooth device...
> 
> ```
> 
> esphome:
>   name: XXXX
>   platform: ESP32
>   board: esp-wrover-kit
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>     
> captive_portal:
> logger:
> api:
>   encryption:
>     key: "XXXX"
>   ap: 
>     ssid: XXXX
>     password: !secret wifi_password
>     ap_timeout: 3min
> ota:
> 
> esp32_ble_tracker:
>   scan_parameters:
>     active: true
> bluetooth_proxy:   
>   active: true
> 
> button:
>   - platform: safe_mode
>     name: ${display_name} (Safe Mode)
> ```

Hello Pickonedev,

Thank you for your infromation, and I search from google, there have a similiar solution
https://www.creatingsmarthome.com/index.php/2023/03/27/guide-bluetooth-proxy-to-home-assistant-using-esphome/

and I haven't process yet, I have a questions, is my ESP32 Dev1 board which is flashed this Switchbot firmware can transfer to ESPHOME, just install ESPHome from HA, and then open ui , Add new device and type SSID, password
then connect the previous ESP32 Dev1 board to USB port and then the ESPHome will flash the board??

If not , please tell me how to flash the ESP32 to match with HA ESP32 Home 

Thanks~~~

pickonedev, 2023-10-16:
> Hello Pickonedev,
>
> Thank you for your infromation, and I search from google, there have a similiar solution [...]

Yes, this is all what I have done. I just reflashed my esp with ESPHome. When you do that, ESPHome will erase the device first, then it will write the new code. Easy. Use Chrome Browser for this, Firefox does not have support for COM ports and even with an addon, you can still get errors...

yauyauwind, 2023-10-16:
> Yes, this is all what I have done. I just reflashed my esp with ESPHome. When you do that, ESPHome will erase the device first, then it will write the new code. Easy. Use Chrome Browser for this, [...]

Hello pickonedev,

Thank you so much, will try this later!

yauyauwind, 2023-10-17:
Hello All,

Just update my status, I use ESPHome BLE Proxy with switchbot work
Just like the previous content, you need to modify config file , board to match with your ESP32 and change the framework
then add ble tracker and bluetooth_proxy to the config

and I have a problem on flash the ESP32, it's stuck on "Preparing Installation"
I googled, and update driver .... press button on the esp32 etc... nothing work

when I want to give us, I switch the broswer from chrome to edge "acutally I check the chrome that can commuicate with ESP32, cause I can screen the log, that's why I didn't alert the problem maybe on the browser"
magically, it's work
For someone who use the chrome and stuck on this "Preparing Installation" over 20mins, then try EDGE
normally this screen may need around 10~20mins , depend on your machine speed, the background is download and compile the firmware.elf

Thanks~

xXREDXIIIIXx, 2023-10-29:
I have now also moved over to ESPHome, relatively easy following the guide posed above by yauyauwind.
I only have contact & motion sensors and 1 light strip so I cant say what else will be found by ESPHome and will be compatible. There has been zero loss in connectivity or delay - in fact I would say it's slightly improved overall, feels more snappy. 
The Switchbot LED strip was not part of this repo so I had no control via home assistant, but no I do - although I'm looking to move to WLED and remove all of my old LIFX runs that just randomly lose internet for no reason...

iz8mbw, 2023-11-02:
Hello.
For what I know If we switch to ESPHome we can see all our Switchbot Bluetooth devices on Home Assistant.
Well, but personally I need to have MQTT messages on my Broker.
This project "SwitchBot-MQTT-BLE-ESP32" will publish MQTT messages as first main goal, after we can "import" data into Home Assistant (or other systems).

Is it possible via ESPHome to have MQTT messages for all entities (sensors) of all Switchbot Bluetooth devices discovered?
So I would like to have:
Switchbot device -> MQTT -> Home Assistant (like I use with "SwitchBot-MQTT-BLE-ESP32"=
and not:
Switchbot device -> Home Assistant -> MQTT

Thanks!

xXREDXIIIIXx, 2023-11-02:
> Hello. For what I know If we switch to ESPHome we can see all our Switchbot Bluetooth devices on Home Assistant. Well, but personally I need to have MQTT messages on my Broker. This project [...]

Why do you need them to be In mqtt?
You will be able see all the enity details and be able to control them via Home Assistant with esphome [where applicable]

Esphome replaces the need for this repository as it is out dates and not maintained. You will have all of the same information you get from this repository via mqtt as a device in HA

iz8mbw, 2023-11-02:
I need/want sensors data on MQTT because I need to "manage" that sensors data on Linux.
I use some shell script.

More I notice that "SwitchBot-MQTT-BLE-ESP32" is more fast (real real-time) than ESPHome.
With "SwitchBot-MQTT-BLE-ESP32" when my Switchbot Meter Plus change values, the values are sent via MQTT in real-time.

These are the temp values via "SwitchBot-MQTT-BLE-ESP32":
[image]

These are the temp values via "ESPHome":
[image]

As you can see, with "SwitchBot-MQTT-BLE-ESP32" we have a better view of all values.

iz8mbw, 2023-11-02:
and, using ESPHome, the charts on Home Assistant are updated every 5 minutes (and the charts have min/average/max values),
Instead, using "SwitchBot-MQTT-BLE-ESP32", the charts on Home Assistant are updated every times there is a new value returned from the sensor.

iz8mbw, 2023-11-18:
Hello all.
For those want to solve the the Waring "`MQTT entity name starts with the device name in your config... this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes...`"
this what I have done and it works for me.

Replace the [line 3154](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3153) from:
`+ "\"name\":\"" + host + " Linkquality\"," +`
to:
`+ "\"name\":\"Linkquality\"," +`

Replace the [line 3156](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3156) from:
`+ "\"uniq_id\":\"switchbotesp_" + host + "_" + wifiMAC.c_str() + "_linkquality\"," +`
to:
`+ "\"uniq_id\":\"switchbotesp_" + wifiMAC.c_str() + "_linkquality\"," +`

Replace the [line 3162](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3162) from:
`+ "\"name\":\"" + host + " Firmware\"," +`
to:
`+ "\"name\":\"Firmware\"," +`

Replace the [line 3165](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3165) from:
`+ "\"uniq_id\":\"switchbotesp_" + host + "_" + wifiMAC.c_str() + "_firmware\"," +`
to:
`+ "\"uniq_id\":\"switchbotesp_" + wifiMAC.c_str() + "_firmware\"," +`

More you need to edit some other code in the "device" part, according on which SwitchBot device you use,
Since I use only for Meters (as SwitchBot devices), I changed only the Meters part.
Scrolling down I arrived to [line 3417](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3417) and I starts the replacements after this line (so do not edit line 3417).

Replace the [line 3423](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3423) from:
`+ "\"name\":\"" + deviceName + " Battery\"," +`
to:
`+ "\"name\":\"" + " Battery\"," +`

Replace the [line 3432](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3432) from:
`+ "\"name\":\"" + deviceName + " Linkquality\"," +`
to:
`+ "\"name\":\"" + " Linkquality\"," +`

Replace the [line 3441](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3441) from:
`+ "\"name\":\"" + deviceName + " Temperature\"," +`
to:
`+ "\"name\":\"" + " Humidity\"," +`

Replace the [line 3451](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino#L3451) from:
`+ "\"name\":\"" + deviceName + " Humidity\"," +`
to:
`+ "\"name\":\"" + " Humidity\"," +`

It works for me without any issues.
Generally you should remove the "deviceName" after `+ "\"name\":\""'.

Hope this can help you.

NewsGuyTor, 2023-12-13:
Thanks to @iz8mbw , I've created a fork that incorporates his changes. I haven't had the chance to test it yet.

https://github.com/NewsGuyTor/SwitchBot-MQTT-BLE-ESP32/blob/patch-1/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino